vittorioexp/Sim800L-Arduino-Library-revised #2: Not working?

jankluiver, 2017-04-25:
Perhaps a personal issue, but I can't get the library to work.

I have a MEGA that talks via SoftwareSerial (10,11) to my SIM800L on 9600 using AT commands. So HW and SW Serial is OK.

I uploaded your library in IDE and tried several examples. It does not work. In the example below it does not even print normal text to monitor.

I really like the functionality you created. So it was a little disappointment that it did not work. Not sure what it is. I have used libraries many times, normally it works (almost) out of the box. 

Any suggestion?

KR, Jan

#include <Sim800L.h>
#include <SoftwareSerial.h>               
#define RX  10
#define TX  11
Sim800L GSM(RX, TX);

void setup(){
  Serial.begin(9600); 
  GSM.begin(9600);     
  Serial.println("GET PRODUCT INFO: ");
  Serial.println(GSM.getProductInfo()); 
}
void loop(){
}

arthurmolina, 2017-05-17:
I made a pull request to fix it.
The change are only in begin method.

VittorioEsposito, 2017-05-19:
Bug fixed. See latest release.

VittorioEsposito, 2017-05-20:
Do it work for you now?

jankluiver, 2017-05-20:
Hello Vittorio,

First of all: thank you (and Arthur) so much for the effort of creating 
the library and addressing my issue.

I wanted to try it last night, but ran out of time. I will try it today 
(probably tonight) and will let you know.

Kind regards,

Jan

jankluiver, 2017-05-20:
Hello Vittorio,

A first update. The (SW) serial port works now.

I still have some problems with the library. Need to investigate further 
what the cause is (possible causes: bad breadboard connections, weak GSM 
signal).

I set both comms op 9600 (in the examples you use 4800 for SIM800L).

_Interim result_
Send SMS: works fine
Info: provides sometimes info, but not consistent
GSM Coordinates: some output, but no coordinates
RTC: comes back with date :  1/1/4 -> default date SIM800L; does not 
use/get the time from GSM.

I have written Arduino code (not in a library) to extract GSM 
coordinates, date, time, battery voltage and battery level from the 
SIM800L. My SIM800L code works fine, but it is part of a long program. 
So I still would like to move the SIM800L functionality to a library to 
make the main code smaller (easier to read). I have no experience with 
writing library code.

I will do some more testing, hopefully tomorrow, and send you the results.

Kind regards,

Jan

VittorioEsposito, 2017-05-20:
Hi Jan,

Thanks for the work of testing and reporting you are doing.

Does it work if you try to get the product info, the operators list, or the operator you are connected to? I'm talking about "getInformations" example sketch. And what do the functions return?

To get GSM coordinates you should first calculate them with "calculateLocation()". Now i try to send the at command to get coords in a different way (see v. 1.1.3). Eventually, try to read the returns of _AT+CIPGSMLOC=1,1\r\n;_.

I took RTC functions from the original version of the library by Cristian Steib (https://github.com/cristiansteib/Sim800l) so we should ask him.

If you want, we can add your code ( to extract GSM coordinates, date, time, battery voltage and battery level from the SIM800L) into the library to replace the old not-working functions.

Kind regards,

Vittorio

jankluiver, 2017-05-21:
Hi Vittorio,

Many thanks for your mail and suggestions.

In some experiments I got the list of telecom operators printed. But 
often there was no response or the text was mutilated (incorrect 
characters in the text). I am living in an area with weak GSM signal. 
That could be a reason. It could also be a bad connection (wiring) on 
the breadboard. Or perhaps a mix or another reason.

I am not a specialist in this, so I am not sure if my code is the best 
way to extract the data from the returned AT answer. But I have used it 
over thouand times and without any error. I am happy to share the code 
if it can be integrated in the library. Personally I very much like the 
battery status function. I use it to show the battery status on a 
display, but it can also be used to report the battery status by SMS.

I am working on something else at the moment, but I will give you 
detailed test feedback and my code. Hopefully later today, otherwise in 
the coming week.

Kind regards,

Jan

jankluiver, 2017-05-21:
Hi Vittorio,

Send SMS is working fine. Which means that all HW and the GSM network 
are OK.

I have problems with all other example programs. GetInformation works a 
little bit. The rest does not.

I added my code in the document. As mentioned, I never created a library 
myself.

For me my current solution works, but it would be nice to hide all the 
code in a library.

Best regards,

Jan

VittorioEsposito, 2017-05-22:
Hi Jan,

Can you please paste here what does "get information" returns?

Where is the document you mentioned, which contains the code that can be added to the library?

Vittorio

jankluiver, 2017-05-22:
Hi Vittorio,

Could you open the Word attachment, or did it not arrive?

This is what is in the Word document (with some screenshots because 
Arduino monitor did not let me copy characters some times):

My own code (location, date, battery) is at the end, with a header.

*Test Report SIM800L Library*

_Test configuration_

SIM800L running on 4.0V

Arduino MEGA 2560

Software Serial RX=10 ; TX=11

Serial comms 9600

_*Library Example 1: Send Sms: works*_

Tested 3x

_*Library Example 2: Get Informations*_

Attempt 2.1:

GET PRODUCT INFO:

+COPS: (2,"T-Mobile NL","TMO N","20416"),(3,"KPN MOBILE","NL 
KPN","20408"),(3,"vodafone","voda NL","20404"),,(0-4),(0-2)

OK

GET OPERATORS LIST:

AT+COPS=?

GET OPERATOR:

Attempt 2.2 (closed and opened monitor window):

Attempt 2.3 (reset on Arduino):

Same as 2.2

Attempt 2.4 (closed and opened monitor window):

Same as 2.2

Attempt 2.5 (disconnected power from SIM800L & Arduino for reset)

_*Example Library 3 : GSM_Coordinates*_

Attempt 3.1:

Attempt 3.2 (close and open monitor window): same as 3.1

Attempt 3.3 (reset button Arduino): same as 3.1

Attempt 3.4 (remove power):

Location Calculated

Location Code: ,1

+CIPGSMLOC:

Longitude: 1

+CIPGSMLOC: 601

OK

Longitude: AT+CIPGSMLOC=1

_*Example Library 4: GetTimeRTC*_

Attempt 4.1:

Date defaults to 1 January 2004 : is default SIM800L???

SIM800L was locked to GSM network

Time: has nothing to do with local time (16:57 at the moment of testing).

Attempt 4.3 After removing power:

Time is seconds after start Arduino.

_*Example Library 5: UpdateRTC with GSM*_

SIM800L removed from power and waited for network lock.

Attempt 5.1: takes a while to respond. Then SIM800L is blinking fast (no 
GSM network) and gives same time as under 4. No sync.

Attempt 5.2: same as above.

*THIS IS HOW I CODED IT IN ARDUINO *

I am used Serial3 (MEGA) for communication.

void getLocation() {

Serial.println("Location is being retrieved from SIM800L. If not printed 
below = failed.");

Serial3.println ("AT+SAPBR=3,1,\"CONTYPE\",\"GPRS\"");

delay(50);

Serial3.println ("AT+SAPBR=3,1,\"APN\",\"smartsites.t-mobile\"");

delay(50);

Serial3.println ("AT+SAPBR=1,1");

delay (50);

Serial3.println ("AT+CIPGSMLOC=1,1");

// answer format: +CIPGSMLOC: 0,4.382148,52.032688,2016/10/28,14:08:16

delay (200);

confirmationSMSlocation = true;

if (!Serial3.find("+CIPGSMLOC: 0,")) {

confirmationSMSlocation = false;

}

//if (confirmationSMSlocation== false) {Serial.println("No +CIPGSMLOC: 
0,");}

if (confirmationSMSlocation == true) {

delay(50);

longitude = Serial3.readStringUntil(',');

delay(50);

//Serial1.read(); //next character is comma, so skip it using this. Does 
not seem to be necessary.

//delay(50);

latitude = Serial3.readStringUntil(',');

//delay(100);

//Serial1.read();

delay(50);

dateGSM = Serial3.readStringUntil(',');

delay(50);

timeGSM = Serial3.readString();

/*

if (Serial.available() >= 8) {

for (int i=0; i<8; i++) {

char timeGSM[i] = Serial.read();

}

*/

delay(50);

Serial.println ("New longitude, latitude, date and time via SIM800L: ");

Serial.println (longitude);

Serial.println (latitude);

Serial.println (dateGSM);

Serial.println (timeGSM); // this one does not work well.

}

}

BATTERY STATUS

Serial3.println ("AT+CBC"); //request battery status. Response format= 
+CBC: 0,76,4016 0 = not charging, 76 = 76%, 4016 = 4016 mV

delay(200); // was 500

confirmationSMSbattery = true;

if (!Serial3.find("+CBC: 0,")) {

confirmationSMSbattery = false;

}

if (confirmationSMSbattery == true) {// when valid battery status is 
available

delay(100);

percentageBattery = Serial3.parseInt();

voltageBattery = Serial3.parseInt();

}

VittorioEsposito, 2017-05-23:
Hi Jan, 

Thank you for sending the code.

I can't understand why there are three Serial3.println ("AT+SAPBR=...); 

Today i'll try to integrate this functions into the library, replacing the code of some functions.

Can you please test if the sleep mode (and functionality mode) functions of the library work?

Vittorio

jankluiver, 2017-05-23:
Hi Vittorio,

My code might not be the best way of handling things.

I send an AT command to the SIM800L and listen to the response. Then I 
used some "tricks" to filter the desired information from the reply 
string. I don't find it a very elegant way, but it was the best way I 
could find on the internet. There might also be a risk that if the 
response format changes (or is different for another part of the world, 
for example location) that it might not work anymore. The code is 
working and reliable at the moment. (Note: I made some minor changes in 
the code I sent by replacing some Dutch comments and variables by 
English; I checked well, but perhaps there is a forgotten translation).

In my larger code I call the time once (to set the RTC), I call the 
battery status every few second (to display on OLED) and the GSM 
location only at the moment I send a SMS.

I did not need the other functionality, so I did not test it. But I will 
test it, probably tonight.

Kind regards,

Jan

VittorioEsposito, 2017-05-24:
I edited the function calculateLocation() adding your AT commands. 
Is this right?

`bool Sim800L::calculateLocation()
{
    /*
    	Type: 1  To get longitude and latitude
    	Cid = 1  Bearer profile identifier refer to AT+SAPBR
    */

   // uint8_t type = 1;
    //uint8_t cid = 1;
	
	//String tmp = "AT+CIPGSMLOC=" + String(type) + "," + String(cid) + "\r\n";
	//this->SoftwareSerial::print(tmp);
	
	this->SoftwareSerial::print("AT+SAPBR=3,1,\"CONTYPE\",\"GPRS\"");
	
	delay(50);
	
	this->SoftwareSerial::print("AT+SAPBR=3,1,\"APN\",\"smartsites.t-mobile\"");
	
	delay(50);

	this->SoftwareSerial::print("AT+SAPBR=1,1");

	delay (50);

	this->SoftwareSerial::print("AT+CIPGSMLOC=1,1");
	
	// answer format: +CIPGSMLOC: 0,4.382148,52.032688,2016/10/28,14:08:16

    String data = _readSerial(20000);

    if (data.indexOf("ER")!=(-1)) return false;

    uint8_t indexOne;
    uint8_t indexTwo;

    indexOne = data.indexOf(":") + 1;
    indexTwo = data.indexOf(",");

    _locationCode = data.substring(indexOne, indexTwo);

    indexOne = data.indexOf(",") + 1;
    indexTwo = data.indexOf(",", indexOne);

    _longitude = data.substring(indexOne, indexTwo);

    indexOne = data.indexOf(",", indexTwo) + 1;
    indexTwo = data.indexOf(",", indexOne);

    _latitude = data.substring(indexOne, indexTwo);

    return true;

}`

jankluiver, 2017-05-24:
Hi Vittorio,

I was unexpectedly very busy yesterday and also today. Still need to do 
a lot of work tonight.

I have seen this kind of coding in the cpp and h files that I looked at, 
but I am not familiar with this. I cannot give a quick answer.

I will look at it tomorrow, together with the testing of the other 
library examples. Sorry for the delay.

Jan

jankluiver, 2017-05-26:
FYI.  Work keeps pooring in. I need to work also this weekend to catch 
up. I hope to answer on Sunday.

Jan

VittorioEsposito, 2017-05-26:
Hi,

Can you send me a picture of your hardware?

Thank you

jankluiver, 2017-05-26:
If you mean the module, it is the same as this one:

https://cristiansteib.github.io/images/sim800l.jpg

VittorioEsposito, 2017-05-26:
I mean the hardware links between sim800l/arduino/battery. Because my module ever returns **ERROR: SIM not inserted**. 
This is what i do.
AT
OK

AT+CFUN?
+CFUN: 1
OK

AT+CPIN?
+CME ERROR: SIM not inserted

jankluiver, 2017-05-26:
I run the SIM800L on a 3.7 Volt 18650 battery (and in test mode via a 
PSU on 4.0 V @ 2A ).

In my operational design I use a Arduino MEGA and use one of the 3 
additional serial ports (Serial3) and I have a level shifter to convert 
between 2.8 and 5 V.

In the test environment I use the S/W serial port (I would not mind to 
use the Serial3, but don't know how to set that in the library) and a 
voltage divider for the input signal of the SIM800L.

Hmm. Perhaps that could be a reason for my unstable response. I use R  = 
10k-10k, so 5V signal is reduced to 2.5V.  I just checked the spec sheet 
<http://wiki.seeedstudio.com/images/4/46/SIM800L_Hardware_Design_V1.00.pdf>. 
High input is between 2.5 and 2.8 V. So 2.5V is on the limit. I will try 
tomorrow R= 10k : 12k, then output will be 2.72V.

I did not experience problems with SIM. I tried two SIMs in the past and 
3 of these modules, all work fine. Only a pity the the RTC battery 
backup is not an output on the board. I need that functionality and have 
now an external RTC (it is not the cost of 1 USD, but it is extra wiring 
and coding).

KR,

Jan

jankluiver, 2017-05-28:
Hello Vittorio,

I did some more tests. I changed the voltage divider to give a little 
higher signal. It did not make a difference.

The Sleeping Functionality is working.

This morning I discovered Tabs in the Arduino IDE. You can move separate 
functions in separate tabs that still belong to the main program. This 
makes the main program more compact. For me that was the main reason 
that I wanted a library. Technically I have all the things that I want.

KR,

Jan

VittorioEsposito, 2017-05-29:
Hi Jan,

I'm glad to know the sleeping functionality is working.

My module still not works, maybe the 3 SIM cards i tried are too recent. When did you buy yours? 

Vittorio.

VittorioEsposito, 2017-06-27:
Hi Jan, 

did the example sketch "GSM_Coordinates.ino" work for you?
I remember i added your modifications about this in the last version of the library.

jankluiver, 2017-07-03:
Apologies for the slow response. I have two SIM800L modules running 
off-site with my own code. They are used in an operational setting and 
can't be used for testing anymore.

My Arduino IDE had so many libraries and projects, that I decided to 
make a fresh start. I re-installed IDE and I loaded your latest library. 
I used a spare SIM800L. It did not work. My home is not a good place to 
test due to poor GSM coverage. The SIM800L does not connect to the GSM 
network, but the test programs I tried (getInformations, SleepMode) did 
not need network contact and halted already earlier. Probably in the 
Serial Communication set up.  I approached the SIM800L via AT serial 
commands. That worked fine. Not sure what the issue is. Not really 
helpful, but that is all I can do for the moment.