vittorioexp/Sim800L-Arduino-Library-revised #4: How can I put SIM800L in Power down mode with this library?

JuanHauara, 2017-05-24:
Hello, if I define "GSM.setFunctionalityMode(0)", the SIM800L will be in Power down mode? 
Is this the power down mode defined in page 60 of the datasheet in which the module consum 50uA?

VittorioEsposito, 2017-05-26:
setSleepMode(true) : Enable slow clock, it is controlled by DTR. When DTR is high, module can enter sleep mode. When DTR changes to low level, module can quit sleep mode.
setSleepMode(false) : Disable slow clock, module will not enter sleep mode.
This function uses AT+CSCLK=0/1 command. I think it won't sleep instantly.

VittorioEsposito, 2017-07-13:
Does it work for you?

berkesandor, 2017-07-29:
I have a GSM module:
https://www.aliexpress.com/item/SIM800L-V2-0-5V-Wireless-GSM-GPRS-MODULE-Quad-Band-W-Antenna-Cable-Cap/32740662773.html?spm=2114.search0104.3.1.OIJpfe&ws_ab_test=searchweb0_0,searchweb201602_3_10152_10065_10151_10068_10171_10084_10083_10119_5370020_10080_10304_10307_10082_10081_10110_10137_10111_10302_10060_10112_10113_10155_10114_10154_10056_10055_10054_10310_10312_10059_100031_10099_10078_10079_10103_10073_10102_10120_10052_10053_10142_10107_10050_10051-10120,searchweb201603_1,ppcSwitch_5&btsid=a7f7c025-6aa0-47d7-83dc-1303b48a4f50&algo_expid=83089329-2e13-4fc9-8316-f143ed2b59af-3&algo_pvid=83089329-2e13-4fc9-8316-f143ed2b59af
Is it possible to put this into sleep?

JuanHauara, 2017-07-29:
Well, I have not tried that because I am using a Mosfet to power down the alimentation of the module. I chose that because I need the lowest possible power consumption.

VittorioEsposito, 2017-07-29:
If you want to put your module in sleep mode use "setSleepMode(true)" method of the library. To wake up it, use "setSleepMode(false)". See Sim800L.cpp file for more details

berkesandor, 2017-07-30:
I would like to receive SMS in sleep mode as well, to awaken the arduino. Therefore disconnecting the GSM module with a mosfet is not good for me.
What i have a module for, the dtr pin does not get out, can it so much to suffocate the module?

AGenchev, 2018-03-12:
@berkesandor: I fail to set it to low power either. I have SIM800L (red PCB). I plan to connect DTR to the +4Vcc of the module, issue  AT+CSCLK=1 and disconnect the TX wire from the serial port because it may leak current towards the MCU. 
If this doesn't work, there's one more option - One guy from UK  has reported that he got it to low power using  CSCLK=2 mode. This mode does not need DTR pin but uses auto wakeup on RX here: 
https://groups.google.com/forum/#!topic/reading-hackspace/8PwIo7imD0k

VittorioEsposito, 2018-03-18:
Thank you @AGenchev

yuliaretno96, 2021-02-08:
> Well, I have not tried that because I am using a Mosfet to power down the alimentation of the module. I chose that because I need the lowest possible power consumption.

@JuanHauara which MOSFET do you use? and is it connected to VCC or GND pin in SIM800L? Thank you

JuanHauara, 2022-05-13:
Hello @yuliaretno96 and sorry for the delay. I am using the Si2301CDS, is channel P MOSFET so I connect it to the VCC pin.