vittorioexp/Sim800L-Arduino-Library-revised #7: GSM Coordinates issue

NadeeshanJayawardana, 2017-06-27:
Hi, Im a student and im new to ardiuno. I have a module SIM800L with me and i'm using your library to Send SMS and Get GSM coordinates. I used sendSms Example in your library, Its working Great.. but when i try GSM_Coordinates Example, i get these outputs in my serial monitor.. Can you explain it and tell me that where am i wrong?

Location Calculated
Location Code: AT+CIPGSML
Longgitude: AT+CIPGSML
Latitude: AT+CIPGSML

 i fixed the board as the instruction 5v to 5v, GND to GND, VDO to none, SIM TXD to RX10, SIMRXD to TX11, GND to GND, RST to none. and i have to admit Sim800L-Arduino-Library-revised is a great library.. thank you very much for it

VittorioEsposito, 2017-07-13:
Hi,
i'm sorry for the response delay but i was busy due to school exams. 
My gsm module has broken, so we should ask @jankluiver (in the other issue page) which became an expert of this library :D.

jankluiver, 2017-07-13:
I already had my SIM800L working before I found the library and in the end I did not implement the library in my project.  You can send SMS, so unit is working. I think we have a different board, because my board does not have a VDO pin. I use 4 V to power the SIM800L (or 3,7V from LiPo battery), because my board does not have a voltage regulator.  5V for power supply is outside the spec of the SIM800L. Perhaps your board has. Regarding coordinates. I could not get it to work in the library. Best first check if the SIM800L is providing data using AT commands. Connect the SIM800L as you have it now, load a simple serial comms program to write commands to SIM800L and read returned output.  The command AT+CIPGSMLOC= 1    should give allo coordinates. For full instruction set : https://www.elecrow.com/download/SIM800%20Series_AT%20Command%20Manual_V1.09.pdf

NadeeshanJayawardana, 2017-07-13:
Mmm I'll try with AT commands and let you guys know. I appreciate it. Thank you very much for your help and support friends. have a nice day both 🙂👍

roboticboyer, 2017-11-25:
Hi Vittorio
I have tried to use the GSM_Coordinates Example, but any data is returned
I think that it's required first to activate the GPRS (and to set the APN) before invoking AT+CIPGSML
Please could you check

VittorioEsposito, 2017-11-26:
Should i write "AT+CGATT=1\r\n" (referring to page 208 of https://www.elecrow.com/download/SIM800%20Series_AT%20Command%20Manual_V1.09.pdf ) before invoking "AT+CIPGSML=1,1\r\n"(line 264 of Sim800L.cpp) ?

roboticboyer, 2017-11-26:
I suggest to take a look at [https://www.elecrow.com/download/SIM800%20Series_GSM%20Location_Application%20Note_V1.01.pdf](https://www.elecrow.com/download/SIM800%20Series_GSM%20Location_Application%20Note_V1.01.pdf)

VittorioEsposito, 2017-11-27:
So should i send **AT+CLBS=1,1** instead of **AT+CIPGSMLOC=1,1** to the sim800 module?

roboticboyer, 2017-11-27:
In this moment I haven't the Sim800 to do a test (I'm working in a collaborative project with my AUG)

But at page 288 of 380 and 366 of 380 of  [https://www.elecrow.com/download/SIM800%20Series_AT%20Command%20Manual_V1.09.pdf](https://www.elecrow.com/download/SIM800%20Series_AT%20Command%20Manual_V1.09.pdf) there is a further example using AT+CIPGSMLOC

Take a look at [http://www.instructables.com/id/How-to-make-a-Mobile-Cellular-Location-Logger-with/](http://www.instructables.com/id/How-to-make-a-Mobile-Cellular-Location-Logger-with/)

I don't know which is the real difference between AT+CIPGSMLOC (GSM Location) and AT+CLBS (Base Station Location)

Do you have a Sim800?
if yes, I suggest to do some tests using a USB to TTL converter and a PC terminal to understand how the Sim800 responds to the AT commands.