vittorioexp/Sim800L-Arduino-Library-revised #10: USSD support?

BdT141, 2017-10-02:
Hi, great library! I was wondering if you've considered adding USSD support for things like checking balances and such?

Regards,
Brandon

VittorioEsposito, 2017-10-03:
Hi, that's a good idea but i don't have so much time to add that new part. You should try to find out the commands that are needed.(https://www.elecrow.com/download/SIM800%20Series_AT%20Command%20Manual_V1.09.pdf)
Let me know.

Vittorio

BdT141, 2017-10-03:
I've done it in Python for Raspberry Pi, but python's string handling methods are WAY more convenient. If I get it working I'll share the code with you.

VittorioEsposito, 2017-10-04:
Thank you, feel free to give your contribution by editing the library with a pull request when you are sure it works (or you can just write here the code so i'll add it to the library).

Vittorio