vittorioexp/Sim800L-Arduino-Library-revised #25: Can it be made to work with ESP32? request / not an issue

mandgeo, 2021-03-31:
I'm searching for a library that can be used with ESP32 and a gsm module (not a real sim900l, but seems to work like it).
Your library seems to make esp talk with my module, but not in the correct way.
Example. getInformation.ino. It pulls some response from the module , but in a fragmented way. get product info prints ATI (the AT command sent to device), operator list prints one word from the response and operator the other word.
I've tried to change readSerial from char by char reading to readStringUntil, but with no succes.
I'm using one of the hardware ports of esp.

Thanks!

vittorioexp, 2021-04-07:
Yes, with a little bit of effort it can be made compatible with ESP32 by defining specific behaviors for that device between include guards within .h and .cpp files. Unfortunately, I don't have much time. Thank you for your interest.

vvzvlad, 2022-05-13:
Yes, I was also planning to use the library with ESP32 and it didn't work. There are a lot of people here waiting for a fix :)

vittorioexp, 2022-10-12:
If you solved the issues feel free to submit it with a pull request